shiguredo/momo #17: RaspberryPiで音声が再生されない

bemarble, 2018-10-16:
下記環境にて、P2Pモードで音声の再生ができませんでした。
視聴環境はchrome69
chrome://webrtc-internal で確認したところ、オーディオ、ビデオともにパケットの受信ができているため、
Raspberry piに使い方に記載されている以外のデバイスの設定が必要な可能性あり。

環境
端末: Raspberry pi 2 Model B
カメラ: HD ProウェブカメラC910
OS: Raspbian 9.4
Momo: momo-18.10.0-rc0_raspbian_armv7

実行コマンド
```
./momo --audio-codec OPUS --audio-bitrate 128 --video-codec VP9 --video-bitrate 500 --resolution VGA --framerate 30 --log-level 4 p2p --port 8080
```

ログ
```
[000:169] [1949] (audio_mixer_manager_alsa_linux.cc:943): Could not find output volume in the mixer.
[000:169] [1949] (adm_helpers.cc:47): Unable to access speaker.
[000:213] [1949] (audio_mixer_manager_alsa_linux.cc:943): Could not find output volume in the mixer.
ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave
[000:276] [1949] (audio_device_alsa_linux.cc:796): unable to open playback device: No such file or directory (-2)
[000:641] [1949] (audio_device_generic.cc:17): BuiltInAECIsAvailable: Not supported on this platform
[000:642] [1949] (audio_device_generic.cc:27): BuiltInAGCIsAvailable: Not supported on this platform
[000:642] [1949] (audio_device_generic.cc:37): BuiltInNSIsAvailable: Not supported on this platform
```

試したこと
https://gist.github.com/voluntas/102de0b94bdac7835168e3f1882906f6

オーディオデバイスの優先度変更
```
$ cat /proc/asound/modules
 0 snd_usb_audio
 1 snd_bcm2835
```

デバイス情報
```
$ arecord -l
**** ハードウェアデバイス CAPTURE のリスト ****
カード 0: C920 [HD Pro Webcam C920], デバイス 0: USB Audio [USB Audio]
  サブデバイス: 1/1
  サブデバイス #0: subdevice #0
```

voluntas, 2018-10-17:
@bemarble おそらくバグのようです。./momo p2p ではなく./momo sora であれば問題なく動作しました。p2p モードのテストが甘かったです ... 。

映像を出さずに --no-video と --audio-codec OPUS で確認をお願いできますでしょうか。これで正常動作を確認しました。

voluntas, 2018-10-17:
@melpon p2p モードだと --no-video で --audio-codec OPUS で --no-video のときのみ音声が配信されて、--video-codec 指定, --audio-codec OPUS の場合には音声が配信されないという問題があるようです。sora の場合は問題なく動作しているので、何が問題なのか今の所不明です ... 。

調査、修正をお願いします。

bemarble, 2018-10-17:
@voluntas 
ご確認ありがとうございます。
p2pモードで --no-video のオプションをつけて起動した場合、音声が配信されました。

voluntas, 2018-10-19:
@bemarble https://github.com/shiguredo/momo/releases/tag/18.10.0-rc2

今回の問題を修正した 18.10.0-rc2 をリリースしました。ご確認頂ければと思います。

voluntas, 2018-10-29:
こちらで動作確認済みなので一度閉じさせてもらいます。もし問題あれば再度開いてください。

closed 2018-10-29 by voluntas