shiguredo/momo #20: momo-18.10.0-rc1_ubuntu_x86_64において、4K配信ができない

messi49, 2018-10-18:
下記環境にて、momoによる4K映像の配信ができなかった。

## 環境
[配信側]
端末: x86
カメラ: LOGICOOL BRIO
OS: Ubuntu 18.04 64bit
Momo: momo-18.10.0-rc1_ubuntu_x86_64

[受信側]
端末: x86
ブラウザ: chrome69
OS: Ubuntu 16.04 64bit
sora-js-sdk: 1.9.3

実行コマンド
```
$ ./momo --no-audio --video-codec H264 --resolution 4K --video-bitrate 30000 sora --auto wss://{sora_url}/signaling Sora
```
or 
```
$ ./momo --no-audio --video-codec H264 --resolution 4K --video-bitrate 30000 p2p --port 8080
```

上記、p2pとsora連携共に解像度を4K指定した場合に、VGA(640×480)程度の解像度になってしまう問題が発生している。FHDでは、問題なく動作する。
※webrtc-internalsにて確認。

[image]

voluntas, 2018-10-18:
@messi49 

この辺の情報はってもらっていいでしょうか？

```
$ v4l2-ctl --list-devices

# video0 だと課程
$ v4l2-ctl -d /dev/video0 --list-formats-ext
```

参考
https://leico.github.io/TechnicalNote/Linux/webcam-usage

messi49, 2018-10-18:
@voluntas ご対応ありがとうございます。下記になります。

```
$ v4l2-ctl --list-devices
Logicool BRIO (usb-0000:01:00.0-1):
	/dev/video0

$ v4l2-ctl -d /dev/video0 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'YUYV'
	Name        : YUYV 4:2:2
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.042s (24.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.133s (7.500 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 160x120
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.042s (24.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.133s (7.500 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 176x144
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.042s (24.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.133s (7.500 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 320x180
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.042s (24.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.133s (7.500 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.042s (24.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.133s (7.500 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 352x288
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.042s (24.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.133s (7.500 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 340x340
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 424x240
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.042s (24.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.133s (7.500 fps)
[... 267 lines cut ...]
		Size: Discrete 640x360
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.042s (24.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.133s (7.500 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.042s (24.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.133s (7.500 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 1920x1080
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.042s (24.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.133s (7.500 fps)
			Interval: Discrete 0.200s (5.000 fps)
```

voluntas, 2018-10-19:
@messi49 こちら、ありがとうございます。いろいろ調べたところ libwebrtc では 4K は非対応ということがわかりました。Chrome の WebRTC では 4K が利用できたので libwebrtc でもできるだろうと思っていたのですが、libwebrtc 側になにか入れなきゃダメみたいです。

将来的には対応していきたいので、一旦こちら pending とさせてください。動作確認をするためのカメラやマシンリソースを開発者達にすぐに用意できないこともあり、今すぐが難しい状況です。

一旦 FHD でご利用いただけると助かります。

v4l2-ctl では 4K で認識できているので libwebrtc 側の問題というのがわかりました。助かりました。

voluntas, 2018-10-19:
https://github.com/shiguredo/momo/issues/21

voluntas, 2018-10-22:
現状報告です。弊社側の環境ですと BRIO が 4K として認識してくれない状況で、ちょっとはまっています。libwebrtc にパッチを当てる方向で進めているのですが、動作確認がとれなくて困っています。

引き続き進めていきます。

voluntas, 2018-10-23:
@messi49 弊社側の Linux 環境でカメラが 4K として認識してくれない状況です。BRIO 側が悪いのか、サーバ側がわるいのかわからないのですが、お手数ですがハードの情報を公開できる範囲でお教えいただけますでしょうか？ (Linux は Ubuntu 18.04 で合わせてあります)

voluntas, 2018-10-24:
@messi49 弊社で試行錯誤したところ、無事 4K で認識しました！お騒がせしました ... 。4K 対応取り組み進めていきます。

messi49, 2018-10-25:
@voluntas お手数をお掛けしますが、どうぞよろしくお願い致します！

voluntas, 2018-10-26:
@messi49 https://github.com/shiguredo/momo/releases/tag/18.10.0-rc4 4K に対応したバージョンを公開しました。libwebrtc 側にパッチを当てることで解決しました。

ロジクールの BRIO 4K で動作確認を行いました。ただ電源的に不安定なのかわからないですが、うまく動かない時があるようです。その時は気軽に issue を立てて頂ければと思います。

確認、よろしくお願いいたします。

messi49, 2018-11-05:
ご対応頂きありがとうございます。
こちらの環境で試しているのですが、4K出力されずVGAで配信されてしまいます。
実行ログを下記に添付いたします。

#### マシン環境
OS: Ubuntu 18.04
Kernel: Linux autoware 4.15.0-36-generic #39-Ubuntu SMP Mon Sep 24 16:19:09 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
CPU: Intel(R) Core(TM) i7-7700 CPU @ 3.60GHz
Mem: 16GB

#### 実行ログ
```
$ ./momo --no-audio --video-codec H264 --resolution 4K --video-bitrate 30000 p2p --port 8080
[OpenH264] this = 0x0x7f92c4000f50, Warning:Change QP Range from(0,51) to (12,42)
[OpenH264] this = 0x0x7f92c4000f50, Warning:Actual input framerate 0.000000 is different from framerate in setting 30.000000, suggest to use other rate control modes
```

#### webrtc_logs_0
[webrtc_logs_0.txt](https://github.com/shiguredo/momo/files/2547392/webrtc_logs_0.txt)

voluntas, 2018-11-05:
@messi49 こちら確認します。momo のバージョンは 18.10.0 で間違いないでしょうか？

voluntas, 2018-11-06:
@messi49 おまたせしてしまい申し訳ありません。確認したところパッケージのビルドミスが判明しました。再度ビルドしてパッケージを更新しました。

お手数ですがパッケージを再度ダウンロードしてご確認をお願いいたします。

messi49, 2018-11-07:
動作確認できました！
ありがとうございます！

closed 2018-11-07 by messi49