shiguredo/momo #60: /dev/video* なしの環境で起動させると落ちる

Hexa, 2019-02-21:
/dev/video* がない環境で momo を起動させると Segmentation fault (core dumped) で落ちます。（momo-19.01.0_ubuntu-18.04_x86_64 で確認）

```
# ls /dev/video*
ls: cannot access '/dev/video*': No such file or directory
# lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.2 LTS
Release:	18.04
Codename:	bionic
# ./momo p2p
ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave
Segmentation fault (core dumped)
```

tnoho, 2019-02-25:
12983f42 で修正しましたので確認をおねがいします。

Hexa, 2019-02-26:
ありがとうございます！
/dev/video* なしの環境で起動させても、落ちないことを確認しました

```
# ls /dev/video*
ls: cannot access '/dev/video*': No such file or directory
# ./momo p2p
ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave
```

closed 2019-02-26 by Hexa