shiguredo/momo #61: MacOSでのmomoのビルドエラー

rabe1028, 2019-03-07:
MacOS mojaveでmomoのビルドを行なっているのですが，以下のエラーでビルドが停止します．
```
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ar: illegal option -- D
usage:  ar -d [-TLsv] archive file ...
	ar -m [-TLsv] archive file ...
	ar -m [-abiTLsv] position archive file ...
	ar -p [-TLsv] archive [file ...]
	ar -q [-cTLsv] archive file ...
	ar -r [-cuTLsv] archive file ...
	ar -r [-abciuTLsv] position archive file ...
	ar -t [-TLsv] archive [file ...]
	ar -x [-ouTLsv] archive [file ...]
make[2]: *** [_build/macos/libwebrtc.a] Error 1
make[1]: *** [momo] Error 2
make: *** [macos] Error 2
```
以下の部分のarコマンドの-Dによってエラーが発生している模様です．
```
# momo 本体のビルド
/Applications/Xcode.app/Contents/Developer/usr/bin/make _build/macos/momo
ar -r -c -s -D _build/macos/libwebrtc.a build/macos/webrtc-build/macos/obj/pc/peerconnection/sdp_utils.o ....
```

melpon, 2019-03-07:
`brew install binutils` を試して、そちらの `ar` コマンドを使ってみて下さい。

rabe1028, 2019-03-07:
binutilsをインストールして，以下のコマンドを実行し，-Dオプションが機能するか確認しましたが，同様のエラーが発生しました．

コマンド & エラー
```
$ ar -r -c -s -D
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ar: illegal option -- D
```

whichで調べると，/usr/bin/arを参照しております．
binutilsでインストールしたarに切り替えができていないものだと思われます．
gnuのarコマンドに切り替える方法をご存知でしたら，教えていただきたいです．

voluntas, 2019-03-07:
@rabe1028 申し訳ないのですが、こちらは Momo に関する issues のため brew のドキュメントを確認お願いします。brew でインストールした binutils にパスを通してもらえばと思います。

rabe1028, 2019-03-07:
ご迷惑おかけしました．
GNUのarコマンドにPathを通して，ビルドを行いましたが，momoのビルド部分で停止しました．
長文になりますが，ビルド時のログを以下に載せます．リンクで失敗している模様です．
ldコマンドはBSD版が動いているのですが，GNU版を用いる方がよろしいでしょうか？
```
$ make macos
make -C macos
source ../../script/docker.sh && \
	export PATH="/Users/ryo/momo/build/macos/webrtc/depot_tools:/usr/local/opt/binutils/bin:/Users/ryo/.cargo/bin:/Users/ryo/.pyenv/shims:/Users/ryo/.pyenv/bin:/Users/ryo/.cargo/bin:/Users/ryo/.pyenv/shims:/Users/ryo/.pyenv/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/usr/local/share/dotnet:~/.dotnet/tools:/Library/Frameworks/Mono.framework/Versions/Current/Commands:/Applications/Wireshark.app/Contents/MacOS:/Applications/Xamarin Workbooks.app/Contents/SharedSupport/path-bin:/Users/ryo/dev/go/bin:/Users/ryo/dev/go/bin" && \
	export GYP_DEFINES="OS=mac target_arch=x64" && \
	prepare_webrtc_macos "" 2a1d802a35ad3d7c2a8371ce0d4263ee2476e81e /Users/ryo/momo/build/macos/webrtc && \
	cd /Users/ryo/momo/build/macos/webrtc/src && \
	patch -p2 < /Users/ryo/momo/build/macos/../../patch/4k.patch && \
	gn gen /Users/ryo/momo/build/macos/webrtc-build/macos --args='target_os="mac" is_debug=false rtc_include_tests=false rtc_build_examples=false rtc_use_h264=false is_component_build=false use_rtti=true' && \
	ninja -C /Users/ryo/momo/build/macos/webrtc-build/macos && \
	ninja -C /Users/ryo/momo/build/macos/webrtc-build/macos \
	  builtin_audio_decoder_factory \
	  native_api \
	  default_codec_factory_objc \
	  peerconnection
HEAD is now at ffb1ffb8 Roll recipe dependencies (trivial).
HEAD is now at ffb1ffb8 Roll recipe dependencies (trivial).
Removing .cipd_bin/
Removing .cipd_client
Removing .cipd_client_cache/
Removing .versions/
Removing metrics.cfg
HEAD is now at 242cd92ba37 [Onion Soup] Move content::MediaStreamSink to Blink
WARNING: Your metrics.cfg file was invalid or nonexistent. A new one will be created.
Syncing projects: 100% (35/35), done.                                                 

________ running '/usr/local/Cellar/python@2/2.7.15_1/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python src/build/mac_toolchain.py' in '/Users/ryo/momo/build/macos/webrtc'
Skipping Mac toolchain installation for mac
Running hooks: 100% (25/25), done.              
HEAD is now at 2a1d802a35 Merge to M73: Use c=IN IP4 <hostname> to support the presence of hostname candidates.
Syncing projects: 100% (35/35), done.                                                 

________ running '/usr/local/Cellar/python@2/2.7.15_1/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python src/build/mac_toolchain.py' in '/Users/ryo/momo/build/macos/webrtc'
Skipping Mac toolchain installation for mac
patching file modules/video_capture/linux/device_info_linux.cc
Done. Made 684 targets from 177 files in 632ms
ninja: Entering directory `/Users/ryo/momo/build/macos/webrtc-build/macos'
ninja: no work to do.
ninja: Entering directory `/Users/ryo/momo/build/macos/webrtc-build/macos'
ninja: no work to do.
make -C .. MOMO_CFLAGS="-O2" PACKAGE_NAME=macos momo
# 依存ライブラリのビルドと取得
cd build/macos/webrtc-build/macos && ninja
ninja: no work to do.
if [ ! -e libs/json-3.4.0/include ]; then git clone --branch v3.4.0 --depth 1 https://github.com/nlohmann/json.git libs/json-3.4.0; fi
if [ ! -e libs/CLI11-1.6.2/include ]; then git clone --branch v1.6.2 --depth 1 https://github.com/CLIUtils/CLI11.git libs/CLI11-1.6.2; fi
# momo 本体のビルド
/Applications/Xcode.app/Contents/Developer/usr/bin/make _build/macos/momo
c++ -o _build/macos/momo _build/macos/src/rtc/objc_codec_factory_helper.o _build/macos/src/util.o _build/macos/src/sora/sora_websocket_client.o _build/macos/src/sora/sora_server.o _build/macos/src/sora/sora_session.o _build/macos/src/signal_listener.o _build/macos/src/watchdog.o _build/macos/src/rtc/observer.o _build/macos/src/rtc/manager.o _build/macos/src/rtc/video_capturer.o _build/macos/src/rtc/device_video_capturer.o _build/macos/src/rtc/connection.o _build/macos/src/p2p/p2p_server.o _build/macos/src/p2p/p2p_session.o _build/macos/src/p2p/p2p_websocket_session.o _build/macos/src/p2p/p2p_connection.o _build/macos/src/main.o _build/macos/src/ws/websocket.o -L_build/macos -lpthread -lwebrtc -ObjC -F/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/System/Library/Frameworks -ldl -framework Foundation -framework AVFoundation -framework CoreServices -framework CoreFoundation -framework AudioUnit -framework AudioToolbox -framework CoreAudio -framework CoreGraphics -framework CoreMedia -framework CoreVideo -framework VideoToolbox -framework AppKit -framework Metal -framework MetalKit -framework OpenGL -L/Users/ryo/momo/build/macos/boost-1.69.0/lib -lboost_filesystem -Lbuild/macos/webrtc-build/macos/obj/third_party/boringssl -lboringssl
ld: warning: ignoring file _build/macos/libwebrtc.a, file was built for archive which is not the architecture being linked (x86_64): _build/macos/libwebrtc.a
ld: warning: text-based stub file /System/Library/Frameworks/AVFoundation.framework/Versions/A/Frameworks/AVFAudio.framework/Versions/A/AVFAudio.tbd and library file /System/Library/Frameworks/AVFoundation.framework/Versions/A/Frameworks/AVFAudio.framework/Versions/A/AVFAudio are out of sync. Falling back to library file for linking.
ld: warning: text-based stub file /System/Library/Frameworks/OpenGL.framework/Versions/A/Libraries/libGL.tbd and library file /System/Library/Frameworks/OpenGL.framework/Versions/A/Libraries/libGL.dylib are out of sync. Falling ba [... cut ...]
      _aes_init_key in libboringssl.a(bcm.o)
  "_vpaes_encrypt", referenced from:
      _aes_init_key in libboringssl.a(bcm.o)
  "_vpaes_set_decrypt_key", referenced from:
      _aes_init_key in libboringssl.a(bcm.o)
  "_vpaes_set_encrypt_key", referenced from:
      _aes_init_key in libboringssl.a(bcm.o)
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [_build/macos/momo] Error 1
make[1]: *** [momo] Error 2
make: *** [macos] Error 2
```

melpon, 2019-03-07:
こちらの環境では、それぞれのコマンドのバージョンは以下のようになりました。 @rabe1028 さんの環境はどうなっているでしょうか？

```
$ ar --version
GNU ar (GNU Binutils) 2.32
Copyright (C) 2019 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) any later version.
This program has absolutely no warranty.

$ c++ -v
Apple LLVM version 10.0.0 (clang-1000.11.45.5)
Target: x86_64-apple-darwin18.2.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin                                                           

$ ld -v
@(#)PROGRAM:ld  PROJECT:ld64-409.12
BUILD 17:47:51 Sep 25 2018
configured to support archs: armv6 armv7 armv7s arm64 i386 x86_64 x86_64h armv6m armv7k armv7m armv7em arm64e arm64_32                                         
LTO support using: LLVM version 10.0.0, (clang-1000.11.45.5) (static support for 21, runtime is 21)                                                            
TAPI support using: Apple TAPI version 10.0.0 (tapi-1000.11.8.2)
```

rabe1028, 2019-03-08:
こちらの環境は以下のようになっております．
```
$ ar --version
GNU ar (GNU Binutils) 2.32
Copyright (C) 2019 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) any later version.
This program has absolutely no warranty.

$ c++ -v
Apple LLVM version 10.0.0 (clang-1000.11.45.5)
Target: x86_64-apple-darwin18.2.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin

$ ld -v
@(#)PROGRAM:ld  PROJECT:ld64-409.12
BUILD 17:47:51 Sep 25 2018
configured to support archs: armv6 armv7 armv7s arm64 i386 x86_64 x86_64h armv6m armv7k armv7m armv7em arm64e arm64_32
LTO support using: LLVM version 10.0.0, (clang-1000.11.45.5) (static support for 21, runtime is 21)
TAPI support using: Apple TAPI version 10.0.0 (tapi-1000.11.8.2)
```

voluntas, 2019-03-13:
@rabe1028 こちら再現する環境と再現しない環境があり、問題解析が難航中です。なにか進展がありましたらコメントさせていただきますのでお待ちいただければと思います。

voluntas, 2019-04-05:
@rabe1028 おまたせしました。develop ブランチにて修正しました。ご確認お願いします。

rabe1028, 2019-04-05:
@voluntas 無事，ビルドでき，P2Pで動作できました．対応ありがとうございます．

voluntas, 2019-04-05:
@rabe1028 確認ありがとうございます！次のリリースには macOS 対応を含める予定です。

closed 2019-04-05 by voluntas