shiguredo/momo #74: raspberry pi B+, raspbian9.9上で killed by SIGABRT

anatawa12, 2019-06-07:
raspberry pi B+(リビジョン番号a020d3)でmomo-19.02.0_raspbian-stretch_armv7を実行するとAbortedと表示されて落ちます。

実行時のコマンドはこちらです。
`./momo --no-audio --video-codec VP8 --video-bitrate 800 --port 8080 p2p`

straceの出力です。
[trace.txt](https://github.com/shiguredo/momo/files/3265373/trace.txt)

voluntas, 2019-06-07:
@anatawa12 OS のバージョン、利用しているカメラの種類などの情報もいただけますでしょうか？

anatawa12, 2019-06-07:
カメラはロジクール製UVCカメラです。
OSは
```
pi@raspberrypi:~ $ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 9.9 (stretch)
Release:	9.9
Codename:	stretch
```
です。

voluntas, 2019-06-07:
@anatawa12 ありがとうございます。ロジクール了解しました。調査してみます。

anatawa12, 2019-06-07:
一緒に持っているPi3 model B V1.2(基盤いわく)では正しく起動できてました。OS情報は少々お待ちください

voluntas, 2019-06-07:
@anatawa12 B と B+ で違いがある ... 感じなんでしょうか。OS 情報ありがとうございます。お待ちしてます。

anatawa12, 2019-06-07:
Model Bの方のOSは8.0のようでした。
Model Bのリビジョン番号は`a22082`です。

```
pi@raspberrypi:~ $ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 8.0 (jessie)
Release:	8.0
Codename:	jessie
```

tnoho, 2019-06-07:
> raspberry pi B+(リビジョン番号a020d3)

リビジョン番号から 3B+ だと理解してますが、相違ないでしょうか。B+ と 3B+ はかなり違うので。

anatawa12, 2019-06-07:
ごめんなさい。 3 入れ忘れていました。

tnoho, 2019-06-07:
ありがとうございます。調べてみます！

anatawa12, 2019-06-07:
必要なこと、ものがあれば教えてください。すぐに用意しますので。

voluntas, 2019-06-09:
@anatawa12 libasound2 は入っていますでしょうか？音声周りの問題と考えています。
また momo 起動時に --log-level=0 を指定して出力されるログを最大にして、その結果をこちらに貼っていただけると助かります。

anatawa12, 2019-06-09:
libasound2に関してはこうなりました。
```
pi@raspberrypi:~ $ sudo apt-get install libasound2
Reading package lists... Done
Building dependency tree
Reading state information... Done
libasound2 is already the newest version (1.1.3-5+rpt4).
libasound2 set to manually installed.
0 upgraded, 0 newly installed, 0 to remove and 7 not upgraded.
```
ですので既にでにインストールされてました。
ログはこの様になりました。
[log.txt](https://github.com/shiguredo/momo/files/3269314/log.txt)

tnoho, 2019-06-09:
通常ですと、このあとに再生デバイスの初期化処理が走るのですがそれが記録されていないので、asound の設定の問題かなと考えております。
`--no-sound` はマイクの音声を送らないオプションでして、再生デバイスは掴んでしまうのです。
再生デバイス周りの設定などログを見る限りでは特段触っているように見受けられないのですが、編集していたりしませんでしょうか。

anatawa12, 2019-06-09:
編集した覚えはないですね。

anatawa12, 2019-06-09:
リセットなどの方法はあるでしょうか？

anatawa12, 2019-06-09:
今確認したところ、`libasound2`は `openjdk-8-jre`の前提としてインストールされてました。
それが関係しますか？

tnoho, 2019-06-09:
そこは関係しないです。もう少し調べてみますので、少々お待ちください。

tnoho, 2019-06-09:
[momo-raspbian_armv7-m75-issue74.tar.gz](https://github.com/shiguredo/momo/files/3269485/momo-raspbian_armv7-m75-issue74.tar.gz)

手元環境が M75 なので M75 になってしまっていますが、このバイナリを試していただけますか？ Audio を全て潰してあります。

anatawa12, 2019-06-09:
通常に起動してhtml表示できました。
(Pi 3 Bの方と同じなのでバグではなく、ブラウザその他の問題だと勝手に思っている)映像がうまく流れないこと以外問題無いです。

tnoho, 2019-06-09:
であれば、あたりですね。正式な対応方針は追って決めます。

映像がうまく流れないというバグは、それはそれで気になりますが。
Chrome で下記のコマンドで接続を試してみてダメそうなら

`./momo --log-level=1 --no-audio p2p`

全部のログを貼った Issue をまた、立てていただければと思います。
video-codec と video-bitrate オプションは P2P モードでは使えません。HTML で VP8 を指定してください。ビットレートはまた違う指定方法があるのですが、今の momo の html には記載していません。行いたい場合は `b=AS` でググるとヒットするかと思います。

anatawa12, 2019-06-09:
#75 で話したとおり映像がうまく流れないことはこちらの原因でした。このissueでのクラッシュする問題は`uv4l`を終了させたあとでもクラッシュします

voluntas, 2019-06-09:
@anatawa12 develop ブランチに libwebrtc M75 対応と、no-audio のときはデバイスを掴まないという仕組みを追加しています。リリースは先なので自前でビルドして動かしていただいても良いですか？

anatawa12, 2019-06-10:
今確認したところ落ちましたね。ビルドをとりあえずやり直したりmac->piの転送の方法など変えて試します

voluntas, 2019-06-10:
@anatawa12 確認した際のコマンドと、トレースログやログの出力などをお願いできますか？こちらでも確認させていただこうと思います。

anatawa12, 2019-06-10:
momo自体は多分問題なかったです。僕が間違ってintel mac上で実行させようとしてしまったのか、ファイルを共有するときのミスかと思います

anatawa12, 2019-06-12:
@voluntas 10日の時点でのdevelopブランチで問題なく実行できました。

voluntas, 2019-06-12:
@anatawa12 無事動作したとのことで了解しました。このブランチは閉じさせていただきます。

closed 2019-06-12 by voluntas