shiguredo/momo #75: 動画が描画されない/送信されない

anatawa12, 2019-06-09:
#74 と同じPi 3 B+からの映像が流れません。
HTMLのプルダウンメニューでVP8を指定しても変わりません。
実行ファイルは #74 のM75のもので、コマンドも #74 のものです。

気づいた点：
- 41秒のSTUN pingがおわる前にICE connection status has changed to completedになる

ログファイルです：
[log-no-video.txt](https://github.com/shiguredo/momo/files/3269508/log-no-video.txt)

tnoho, 2019-06-09:
うおー。。。カメラ何使ってます？ Logicool の C930 ですか？

anatawa12, 2019-06-09:
C270です

anatawa12, 2019-06-09:
LogicoolのC270です

tnoho, 2019-06-09:
いつごろ購入された C270 でしょうか？
手持ちのものと全然サポートフォーマット情報がちがいまして。。。念の為 lsusb コマンドの結果もいただけると嬉しいです。

anatawa12, 2019-06-09:
lsusb:
```
pi@raspberrypi:~/momo-2 $ lsusb
Bus 001 Device 004: ID 046d:0825 Logitech, Inc. Webcam C270
Bus 001 Device 005: ID 0424:7800 Standard Microsystems Corp.
Bus 001 Device 003: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 002: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
購入時期：2018/09です

anatawa12, 2019-06-09:
接続機器はUSBカメラと何もつないでない(キーボードを繋ぐ用の)USBケーブルです

tnoho, 2019-06-09:
[momo-raspbian_armv7-m75-issue75.tar.gz](https://github.com/shiguredo/momo/files/3269529/momo-raspbian_armv7-m75-issue75.tar.gz)

CPU使用率が上がってしまいますが、これを試してもらえますか？

voluntas, 2019-06-09:
@anatawa12 念のためですが確認しているブラウザと環境 (OS) を教えていただけますか？

# このあたりテンプレート化しておくべきでした、矢継ぎ早に質問して申し訳ないです

anatawa12, 2019-06-09:
ブラウザなどどこらへんまで必要化を理解してないので質問していただこうと思っていました。こちらこそすみません。

name | value
-- | --
Google Chrome | 74.0.3729.169 (Official Build) (64-bit)
Revision | 78e4f8db3ce38f6c26cf56eed7ae9b331fc67ada-refs/branch-heads/3729@{#1013}
OS | Mac OS X
JavaScript | V8 7.4.288.28
Flash | 32.0.0.192 /Users/anatawa12/Library/Application Support/Google/Chrome/PepperFlash/32.0.0.192/PepperFlashPlayer.plugin
User Agent | Mozilla/5.0 (Macintosh; Intel Mac OS X 10_14_5) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/74.0.3729.169 Safari/537.36
Command Line | /Applications/Google Chrome.app/Contents/MacOS/Google Chrome -psn_0_127007 --flag-switches-begin --enable-devtools-experiments --flag-switches-end --restore-last-session

voluntas, 2019-06-09:
@anatawa12 ありがとうございます。最新版は Chrome M75 なので上げてもらったほうがいいかもしれません。（本件とは別ですのでスルーしてもらっても結構です）

念の為ですが、カメラは macOS で正常に認識しますでしょうか？

以下で簡単に確認できますので、お時間あるときに確認をお願いします。
https://test.webrtc.org/

anatawa12, 2019-06-09:
過去にiMacで接続できたこと、uv4lで接続できたことを確認しました。今からもう一度確認します

anatawa12, 2019-06-09:
今でも使えることを確認しました

voluntas, 2019-06-09:
@anatawa12 確認ありがとうございます。次は @tnoho お送りしたバイナリでの確認をお願いできればと思います。

https://github.com/shiguredo/momo/issues/75#issuecomment-500209628

anatawa12, 2019-06-09:
@tnoho @voluntas いま原因がわかりました。`uv4l`のサーバーの自動起動が切れてなかったようでした。
先に`uv4l`がUSBカメラを使用してたから表示されなかったと思われます。`uv4l`を終了させたらうまくいきました。すみません。

closed 2019-06-09 by anatawa12

voluntas, 2019-06-09:
@anatawa12 おぉー、なるほど。むしろありがとうございます。無事表示できてよかったです。

また何か問題あったら気軽に Issue 立ててください。