shiguredo/momo #86: ngrok経由での接続が突然切断される

rabe1028, 2019-07-17:
**発生している問題**
momoを外部から接続できるようにするため，ngrokを用いて接続していたのですが，５分くらい経つとiceconnectionstateがdisconnectになり，映像が途切れてしまう問題が発生しています．

**再現する環境**
Macbook Pro (15-inch, 2018)
Ubuntu 18.04 LTS

```
momo --port 8080 --log-level 1 p2p
```

**想定される原因**

wiresharkでパケットを見ていたところ，突然ICMPのDestination unreachable (Port unreachable)が返ってきていました．そのため，ホールパンチしていたポートが閉じられたことが原因だと思われます．

**カメラ/マイク**

Mac : Macbook Proに搭載されているwebcam
Ubuntu : v4l2loopbackを用いた映像

**自前バイナリ/提供バイナリ**

Mac : 3bfe4393010ec18d6d519f33ab3dad1854a535e3
Ubuntu : eeac602be85029512c1515603dde8c898f4350b6

**--log-level 1 の出力ログ**

macbookで取得したものです．
[momo-log.txt](https://github.com/shiguredo/momo/files/3400008/momo-log.txt)

voluntas, 2019-07-17:
@rabe1028 ngrok ではなく ayame を使って確認をしてもらってどんな感じか教えてもらえますか？

rabe1028, 2019-07-19:
ayameとmomoの接続がうまくいっていないため，もう少し時間がかかりそうです．
進展がありましたら，またご連絡します．

voluntas, 2019-07-19:
@rabe1028 TURN サーバが必要な環境な場合は ayame では現時点では繋がりません。

ネットワーク的な問題の可能性が高いので https://test.webrtc.org/ で Connectivity を確認していただければと思います。

voluntas, 2019-08-03:
@rabe1028 こちら 2 週間過ぎてますので、一旦 close させていただきます。何かありましたら別途新規で issue を立ててください。

closed 2019-08-03 by voluntas