shiguredo/momo #94: usrvcsm,mmalなどでallocationが失敗したログが出力された後にSEGVで終了する

yuroyoro, 2019-08-14:
**バグを説明してください**

momoプロセスがSEGVで終了する

**再現する**

systemdのサービスとして以下のコマンドで起動する

```
/home/pi/momo/momo --no-audio --log-level=1 ayame wss://ayame-lite.shiguredo.jp/signaling yuroyoro@bumbun-blau --signaling-key $SIGNALING_KEY
```

**予想される行動**

プロセスが終了しない

**カメラ/マイク**

`Raspberry Pi Camera V2 / 913-2664`

lsusb のコマンド結果

```
Bus 001 Device 004: ID 0424:7800 Standard Microsystems Corp.
Bus 001 Device 003: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 002: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

**自前バイナリ/提供バイナリ**

Githubのリリースページから以下のtarをダウンロードして利用

`momo-19.08.0-rc0_raspbian-buster_armv7.tar.gz`

**--log-level 1 の出力ログ**

`--log-level=3` ですが添付します

[momo-journal-log.txt](https://github.com/shiguredo/momo/files/3501646/momo-journal-log.txt)

**追加の情報**

この問題に関する他の情報をここに追加してください。

yuroyoro, 2019-08-15:
[mono-log.txt](https://github.com/shiguredo/momo/files/3504779/mono-log.txt)
`--log-level 1 ` で再現したので添付します。
iOS safariから接続しようとしたら発生しました

voluntas, 2019-08-15:
@yuroyoro ありがとうございます！ H.264 のハードウェアエンコーダー周りのメモリーリークで、修正はできそうなんですが、他のところに影響が出そうなので解決まではもう少しかかりそうです。

tnoho, 2019-08-15:
@yuroyoro 報告ありがとうございます。GPU の RAM を食いつぶしてしまう不具合でした。　
bcc71dfa で修正をしてみました。試験用のパッケージを作成しましたので、ご確認いただければ幸いです。
[momo-19.08.0-rc0_raspbian-buster_armv6.tar.gz](https://github.com/shiguredo/momo/files/3505598/momo-19.08.0-rc0_raspbian-buster_armv6.tar.gz)

tnoho, 2019-08-15:
修正版を再掲します

[momo-19.08.0-rc0_raspbian-buster_armv6.tar.gz](https://github.com/shiguredo/momo/files/3506190/momo-19.08.0-rc0_raspbian-buster_armv6.tar.gz)
[momo-19.08.0-rc0_raspbian-buster_armv7.tar.gz](https://github.com/shiguredo/momo/files/3506191/momo-19.08.0-rc0_raspbian-buster_armv7.tar.gz)

voluntas, 2019-08-16:
@yuroyoro 最新版をビルドして公開しておいたので、こちらを使ってみてください。

https://github.com/shiguredo/momo/releases/tag/19.08.0-rc1

voluntas, 2019-08-22:
@yuroyoro こちら、ロングラン試験して問題ないことをこちらで確認しました。また問題ありましたら reopen または、新しく open してください。

今回はご報告本当にありがとうございます。助かりました。

closed 2019-08-22 by voluntas