shiguredo/momo #100: ローカルネット上のAyameを使用して接続した場合にSEGVが発生する

roy-n-roy, 2019-09-24:
**バグを説明してください**

Momoを起動して、ローカルホスト上のAyameに接続し、
`register`メッセージを送信後に受信した`accept`メッセージ内に、
IceServersが含まれていない場合、SEGVが発生する。

**再現する**

再現時コマンド
./momo --no-audio --video-device /dev/video0 --force-i420 --resolution FHD --fixed-resolution --log-level 0 ayame ws://localhost:3000/signaling test >& momo_log.txt

**予想される行動**

プロセスが終了しない。
AyameからIceServersが返答されない場合、
"stun:stun.l.google.com:19302"を使用して処理が継続される。

**カメラ/マイク**

カメラ: [auvidea B101 HDMI to CSI-2 Bridge (15 pin FPC)](https://auvidea.eu/b101-hdmi-to-csi-2-bridge-15-pin-fpc/)
マイク: なし

lsusb のコマンド結果を貼ってください。
`pi@raspberrypi:~ $ lsusb`
`Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`
`Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub`
`Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub`
`Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`

**自前バイナリ/提供バイナリ**

提供バイナリ
momo-19.09.0_raspbian-buster_armv7.tar.gz に含まれる `momo`

**--log-level 1 の出力ログ**

[momo-log-level-1.txt](https://github.com/shiguredo/momo/files/3645202/momo-log-level-1.txt)

**追加の情報**

coredumpファイルも添付いたします。
[core.zip](https://github.com/shiguredo/momo/files/3645283/core.zip)

自前ビルド[commit 7ce0bc7894ad5df2a0b8876d73f4f8f331e38ff4](https://github.com/shiguredo/momo/commit/7ce0bc7894ad5df2a0b8876d73f4f8f331e38ff4)でもデバッグ実行したところ、再現しました。

IceServersが含まれていない場合は [src/ayame/ayame_websocket_client.cpp:307](https://github.com/shiguredo/momo/blob/ffe1c0eaad2473ba577b16ac6d965d24e175efaf/src/ayame/ayame_websocket_client.cpp#L307) に
落ちることが想定されていると思われますが、
想定している`json::type_error`が発生せず、`ice_servers_`が設定されないため、
[src/rtc/connection.cpp:18](https://github.com/shiguredo/momo/blob/ffe1c0eaad2473ba577b16ac6d965d24e175efaf/src/rtc/connection.cpp#L18) でSEGVが発生しているようです。

voluntas, 2019-09-25:
@roy-n-roy e2ae19717a05112c4d4de8a9687965a4f3e06146 こちらで修正済みです。

19.09.2 はもう少ししたらリリースしますのでお待ち下さい。

voluntas, 2019-09-25:
@roy-n-roy 上記バグを修正した 19.09.2 をリリースしました。ご確認、お願いします。

https://github.com/shiguredo/momo/releases/tag/19.09.2

roy-n-roy, 2019-09-25:
@voluntas 正常動作を確認出来ました。ありがとうございます。

closed 2019-09-25 by roy-n-roy