shiguredo/momo #103: Jetson nanoで4Kの配信をするときにフレームレートが上がらない

tetsu-koba, 2019-10-09:
**バグを説明してください**

Jetson nanoで4Kの配信をするときにフレームレートを30fps に指定しても実際には13fpsくらいまでしか上がらない。

**再現方法**

```
#!/bin/sh
AUDIO_OPT='--disable-echo-cancellation --disable-auto-gain-control --disable-noise-suppression --disable-highpass-filter --disable-typing-detection'

./momo $AUDIO_OPT --log-level 1 --video-device /dev/video0 --resolution 4K --fixed-resolution --framerate 30 test 
```

--use-native オプションをつけるとOSごとハングアップしてしまうのでつけていない。

**予想される行動**

Jetson nanoで4Kの配信をするときにフレームレートが30fps にできる。

**カメラ/マイク**

利用しているカメラやマイクの型番を教えてください。

Logicool BRIO ULTRA HD PRO WEBCAM

lsusb のコマンド結果を貼ってください。

```
Bus 002 Device 005: ID 046d:085e Logitech, Inc. 
Bus 002 Device 002: ID 0bda:0411 Realtek Semiconductor Corp. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 0bda:5411 Realtek Semiconductor Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

**自前バイナリ/提供バイナリ**

提供バイナリ
momo-19.09.2_ubuntu-18.04_armv8_jetson_nano

**--log-level 1 の出力ログ**

[p2p.log](https://github.com/shiguredo/momo/files/3706902/p2p.log)

**追加の情報**

このWebcamからはMJPEGを30fps で取り出せています。

```
 $ v4l2-ctl --set-fmt-video=width=3840,height=2160,pixelformat=1
 $ v4l2-ctl --stream-mmap=3 --stream-count=300
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 30.09 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 30.00 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 30.03 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 30.00 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 30.00 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 30.01 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 30.01 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 30.00 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< 30.00 fps
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
```

tnoho, 2019-10-09:
報告ありがとうございます。

### 実行時のコマンドについて

`--fixed-resolution` を外してみてください。4Kの時には `--fixed-resolution` オプションを使うとレートが安定しない傾向があります。

### --use-native オプション利用時のハングアップについて

Jetson Nano のライブラリにバグがあるため、 `/usr/lib/aarch64-linux-gnu/tegra/libnvjpeg.so` を下記の記事で配布されているものに置き換えてください
https://devtalk.nvidia.com/default/topic/1050162/jetson-nano/r32-1-0-mmapi-and-decodetofd-leak-memory-/

なお、こちらについては Jetson Nano の最新化を行うことで改善するとも聞いています。

### フレームレートが出ない場合

一番多いのは暗い場所で利用しているパターンです。カメラが自動的に露光時間を伸ばすためフレームレートが下がります。部屋を明るくする。もしくはカメラの設定変更が可能な場合はフレームレート優先設定に変更してください。

### Logicool BRIO をご利用の場合

Logicool BRIO ULTRA HD PRO WEBCAM は所有していないためフレームレートが出ない原因については `--fixed-resolution` を外して `libnvjpeg.so` を差し替えたうえで、 `--use-native` をつけて実行して試してもらえればと思います。

### [IMX317を搭載した推奨カメラ](https://ja.aliexpress.com/item/32999909513.html) をご利用の場合

> v4l2-ctl --set-ctrl=exposure_auto=1

を実行してカメラの設定を変更してください。 4K 30fps が出力可能な設定は下記のとおりです

```
$ v4l2-ctl --list-ctrls
                     brightness 0x00980900 (int)    : min=-64 max=64 step=1 default=0 value=0
                       contrast 0x00980901 (int)    : min=0 max=95 step=1 default=1 value=1
                     saturation 0x00980902 (int)    : min=0 max=100 step=1 default=60 value=60
                            hue 0x00980903 (int)    : min=-2000 max=2000 step=1 default=0 value=0
 white_balance_temperature_auto 0x0098090c (bool)   : default=1 value=1
                          gamma 0x00980910 (int)    : min=64 max=300 step=1 default=100 value=100
                           gain 0x00980913 (int)    : min=0 max=255 step=1 default=100 value=100
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=1 value=1
      white_balance_temperature 0x0098091a (int)    : min=2800 max=6500 step=1 default=4600 value=4600 flags=inactive
                      sharpness 0x0098091b (int)    : min=0 max=7 step=1 default=0 value=0
         backlight_compensation 0x0098091c (int)    : min=0 max=100 step=1 default=64 value=64
                  exposure_auto 0x009a0901 (menu)   : min=0 max=3 default=3 value=1
              exposure_absolute 0x009a0902 (int)    : min=1 max=10000 step=1 default=156 value=156
error 5 getting ext_ctrl Pan (Absolute)
error 5 getting ext_ctrl Tilt (Absolute)
                 focus_absolute 0x009a090a (int)    : min=0 max=1023 step=1 default=0 value=0 flags=inactive
                     focus_auto 0x009a090c (bool)   : default=1 value=1
error 5 getting ext_ctrl Zoom, Absolute
```

以上、よろしくお願いします。

tetsu-koba, 2019-10-10:
回答ありがとうございます。

`libnvjpeg.so` を差し替えると、`--use-natie` の有無に関わらず接続したときに以下のエラーメッセージでmomoが終了してしまうようになってしまいました。

```
JPEG parameter struct mismatch: library thinks size is 760, caller expects 776
```

他のライブラリとバージョンが合わないようです。
jetson nano のSDカードのイメージにはこのファイルを使っています。
jetson-nano-sd-r32.2.1.zip

tnoho, 2019-10-10:
R32.2 の場合はパッチが異なるようです。

https://devtalk.nvidia.com/default/topic/1060896/jetson-tx2/jetpack-4-2-1-nvjpeg-leaking/

こちらで配布されているパッチを適用して再度試していただけますでしょうか。

tetsu-koba, 2019-10-11:
ありがとうございます。このパッチで4K 30fps  できました！！

まとめ。
jetson nano のSDカードのイメージは` jetson-nano-sd-r32.2.1.zip`
`/usr/lib/aarch64-linux-gnu/tegra/libnvjpeg.so` を
https://devtalk.nvidia.com/default/topic/1060896/jetson-tx2/jetpack-4-2-1-nvjpeg-leaking/
で配布されている `R32_2_libnvjpeg.zip` のものに差し替える。
p2pの開発者モードを使用する場合に実行するコマンドは以下。

```
#!/bin/sh
AUDIO_OPT='--disable-echo-cancellation --disable-auto-gain-control --disable-noise-suppression --disable-highpass-filter --disable-typing-detection'

./momo $AUDIO_OPT --log-level 2 --video-device /dev/video0 --use-native --resolution 4K --framerate 30 test 
```

chromeブラウザで `http://koba-jet.local:8080/html/test.html` を開く。(jetson nanoのホスト名がkoba-jet の場合)
chromeブラウザの`chrome://webrtc-internals/` で解像度とフレームレートを確認する。

closed 2019-10-11 by tetsu-koba

voluntas, 2019-10-11:
のちほどドキュメントに追記しておきます！

voluntas, 2019-10-11:
https://github.com/shiguredo/momo/blob/develop/doc/USE_JETSON_NANO.md ドキュメントに追記