shiguredo/momo #107: The resource '/' was not found.

blimjoe, 2019-10-19:
Hi,
I'm trying to deploy momo on Ubuntu 18.04 on x86_64. Everything is alright during compiling. 
I execute '$ ./momo test' on my machine and use another PC to visit "http://ip:8080", the result is "The resource '/' was not found. ". But the camera on PC is on. Could you please give some suggestion?
I cannot figure out why. I don't know whether I have done something wrong or not. I am so sorry that I cannot read Japanese. I can only use google translation to read the documents. If the document can be translate to English, it would be much better. 
Thanks!

voluntas, 2019-10-21:
Thank you for using it. However, we do not respond in English. This time, we will respond in English.

Https://github.com/shiguredo/momo/blob/develop/doc/USE_TEST.md
```
http://[momo の IP アドレス]:8080/html/test.html にアクセスして接続してみてください。
```

Try accessing "/html/test.html" instead of "/".

If this doesn't work, please tell me the build procedure. It is written assuming that "build / make ubuntu-18.04_x86_64" is executed.

voluntas, 2019-10-26:
This issue is closed because there is no response for a week.

closed 2019-10-26 by voluntas

sounance, 2020-08-13:
If you build momo from source, you need to copy the contents of the /html folder to the same folder as the target momo.