shiguredo/momo #110: Ubuntu環境において、ROS対応バイナリのビルドが失敗する

TE-TakuyaSugitani, 2019-11-07:
**バグを説明してください**

Ubuntu + docker環境において、ROS対応バイナリのビルドがエラーで失敗してしまいます

**再現する**

ビルド環境は以下の通りです
```
Ubuntu 18.04.3 LTS
Docker version 19.03.3., build a872fc2f86
```
※バイナリのビルド環境はUbuntu 18.04ですが、実行は別環境で行う予定でした

USE_ROS.md を参考に、以下のコマンドを実行しました
```
make ubuntu-16.04_x86_64_ros.clean
make ubuntu-16.04_x86_64_ros
```

**予想される行動**

ビルドが完了し、momoのバイナリが作成される

**自前バイナリ/提供バイナリ**

以下のコミットハッシュ時点のコードをビルドしています
```
b5c3fa7e738ca7576ee4c794c6500ccbeb8c029c
```

**ビルドログ**

ビルドログ中のエラーと思われる箇所は以下の通りです
[momo_ros_x86_error.txt](https://github.com/shiguredo/momo/files/3817939/momo_ros_x86_error.txt)

voluntas, 2019-11-07:
@TE-TakuyaSugitani 報告ありがとうございます。ビルドが通らないことこちらでも確認できました。調査進めていきます。

voluntas, 2019-11-08:
@TE-TakuyaSugitani f628611035f978ed478ab70c49a78dddfd971d6c の最新の develop にて修正しましたので、ご確認いただけますか？

TE-TakuyaSugitani, 2019-11-08:
@voluntas 
無事ビルドが通ることを確認しました。ご対応ありがとうございます！

closed 2019-11-08 by voluntas