shiguredo/momo #124: jetson nano用momoで 1920x1080の配信をしたときに画面下端に緑の帯がでる

tetsu-koba, 2019-12-24:
**バグを説明してください**

jetson nano用momoで 1920x1080の配信をしたときに画面下端に緑の帯がでる
3840x2160 や 1280x720 のときには発生しない。
画面解像度はWebRTC downstream側の chrome の chrome://webrtc-internals で確認した。

**再現する**

```
SIGNALING_URL=...省略
CHANNEL_ID=...省略
AUDIO_OPT='--disable-echo-cancellation --disable-auto-gain-control --disable-noise-suppression --disable-highpass-filter --disable-typing-detection'

../current/momo $AUDIO_OPT --log-level 2 --video-device /dev/video0 --use-native --resolution FHD --framerate 30 sora --auto --video-codec H264 --audio-codec OPUS --audio-bitrate 96 $SIGNALING_URL $CHANNEL_ID
```

**予想される行動**

画面下端に緑の帯がでない

**カメラ/マイク**

IMX317 の4Kカメラと ロジクール C922 の2つのカメラで確認した。

**自前バイナリ/提供バイナリ**

提供バイナリ
momo-19.12.1_ubuntu-18.04_armv8_jetson_nano

**--log-level 1 の出力ログ**

[webrtc_logs_0.txt](https://github.com/shiguredo/momo/files/3997920/webrtc_logs_0.txt)

**追加の情報**

スクリーンショット
 --resolution を FHD (1920x1080) のときに下端の緑の帯がでる。
HD (1280x720) のときにはこの帯はでない。

[image]
[image]

tetsu-koba, 2019-12-24:
H.264は 16x16 のブロックに分割してエンコードされます。
2160, 720 は16の倍数ですが、1080 は16の倍数でないということが関係していそうです。

tetsu-koba, 2019-12-25:
`--use-native` オプションをつけないで実行してみたら1920x1080でも緑の帯は出ずにただしく表示できました！どうやらH.264のエンコードではなくて、jpegのデコードのところの問題のようです。
`--use-native` なしでも1920x1080は30fpsでます。
しかし、4K では`--use-native` なしではすぐに映像が止まってしまうので`--use-native`が必須のようです。

voluntas, 2019-12-25:
@tetsu-koba おっしゃるとおり MotionJPEG のハードウェアデコーダー問題っぽいですね。

msnoigrs, 2019-12-25:
私の環境でも、緑の線が出ることを確認しました。--use-native指定あり/FHD/H264の条件で再現します。--use-native指定なしでは同様に緑の線は出ません。jpegのHWデコーダーとH.264のHWエンコーダーとの組み合わせにおいて起きる現象のように見えます。@tetsu-koba さんのおっしゃる通り、16の倍数に関係しそうです（私はH.264の16x16の仕様については、今回初めて知りました、勉強になります）。

tnoho, 2020-02-14:
再現しました。修正したいと思います。

tnoho, 2020-02-14:
@tetsu-koba 遅くなりましたが、修正が完了しましたのでご確認ください。

お察しの通りjpegデコードの問題でした。
入力高さが 1080 のときにjpegデコーダが16の倍数の 1088 で出力して緑の帯が発生し、中間に入っているサイズ変換が1088から1080に縮小処理を行っていたようです。
そのため、出力映像は高さが1080のまま緑の帯が発生することとなっていました。

サイズ変換時に高さを 1080 で切り取ることにより緑の帯は発生しなくなっています。

ご報告ありがとうございました。確認できましたら、クローズいただければと思います。

tetsu-koba, 2020-02-21:
すみません。通知を見逃していました。
機材が使える時に動作確認して報告します。

voluntas, 2020-02-22:
@tetsu-koba ありがとうございます、お時間あるときでいいので確認してもらえるとありがたいです。

tetsu-koba, 2020-02-25:
確認できました。対応どうもありがとうございました！
本件クローズします。

closed 2020-02-25 by tetsu-koba