shiguredo/momo #127: sora モードの multistream + sendonly で長時間利用してる際に他の sendrecv が退出した時にセグメンテーションフォルトで落ちる

voluntas, 2020-01-11:
この issue はこちらからの引き継ぎです https://github.com/shiguredo/momo/issues/119

**バグを説明してください**

sora モードの sendonly + multistream で接続している際、sendrecv / recvonly で参加したあとに離れると momo がセグメンテーションフォルトで落ちる

**再現する**

```
AUDIO_OPT='--disable-echo-cancellation --disable-auto-gain-control --disable-noise-suppression --disable-highpass-filter --disable-typing-detection'
SIGNALING_URL=wss://....
CHANNEL_ID=566ce1...

../current/momo $AUDIO_OPT --video-device /dev/video0 --log-level 4 --resolution QVGA --framerate 20 sora --role sendonly --auto --video-codec H264 --video-bitrate 1000 --audio-codec OPUS $SIGNALING_URL $CHANNEL_ID --multistream
```

**予想される行動**

sendonly では SDP 交換は発生しないため、それ以外が要因と思われる。

**カメラ/マイク**

カメラは ロジクール C270

**自前バイナリ/提供バイナリ**

最新版ではないことに注意

momo-19.12.0_raspbian-buster_armv7

**追加の情報**

https://github.com/shiguredo/momo/issues/119

voluntas, 2020-01-11:
@tetsu-koba から頂いたログを見た限り「長時間接続」ではなく「再接続時」に失敗しているようにみえる。まずは再現を試みる。

tetsu-koba, 2020-01-11:
Sora Labo を相手にして同じ手順を試してみてよいですか。

voluntas, 2020-01-11:
@tetsu-koba はい、是非お願いしたいです。

tetsu-koba, 2020-01-12:
DTLSのエラーが多発して映像がでない状態になったけど、そのあとうまく再接続して復活するというログがとれました。
エラーが発生したのは 2020/01/12 16:02 頃です。
[log_ok.zip](https://github.com/shiguredo/momo/files/4050443/log_ok.zip)

voluntas, 2020-01-12:
@tetsu-koba 大変たすかります。

Sora Labo の Sora のログを見たところ、この切断は「クライアント」から行われているようです。Sora から何かを送ったトリガーで何かしら新規接続と勘違いしてつないでるような挙動なのがわかりました。

- Sora が multistream 離脱時に余計な情報を sendonly に送っている
- Momo が multistream 離脱時に送られてくるもので余計なことをしている

で、 Momo 側がなにか勘違いして DTLS を最初から始める（つまり接続のやり直し) というところまでは絞れました。

voluntas, 2020-01-12:
クライアントが切断しているのはおそらく DTLS を試みたが、うまく行かないからというところでしょう。そもそも挙動として Sora 側は Signaling がつながった状態で DTLS の再ハンドシェイクはセキュリティの観点から実装していないため、 Momo が期待した動きになっていないと考えています。

再ハンドシェイクの実装自体する気はないので、そもそも再ハンドシェイクが走るのはおかしい気がしています。実際に DTLS パケットが 1 発だけ送られているのを Sora のログから確認できました。

voluntas, 2020-01-12:
@tetsu-koba おそらくこれだろう！という問題は見つけられたので、ちょっと Sora/Momo 両方の点からソースコードのチェックをします。大変助かりました。

セグフォに関しては再現しない感じでしょうか？

tetsu-koba, 2020-01-12:
状況が絞れてよかったです！

SEGVについてはログを送ったものしか発生していないです。
そのときのログにcore file も一緒に送っているので、そちらでシンボル情報付きの実行ファイルとcore file をgdb にかければSEGV発生時のスタックトレースが見られると思うのですがどうでしょうか。

voluntas, 2020-01-12:
@tetsu-koba 了解しました！ セグフォするときとしないときがあるのは気持ち悪いですね。

ちょっとログを見ていたら、すごい気になるのがありました。おっしゃっていた sendrecv (upstream) が離脱した時に DTLS が送られてきているのはあるんですが、特に何もしていないときも送っているみたいです。何かしら理由がありそうなので、調べてみます。

voluntas, 2020-01-12:
@tetsu-koba 怪しそうな部分を一つ見つけました。multi + sendonly に余計な SDP を送っていることがわかったので、それを修正した Sora を Sora Labo にデプロイしてみました。お時間ある時、確認していただけると助かります。

torikizi, 2020-01-13:
修正反映後のSora Laboにて私の環境で確認をしてみました。

### まとめ
再現しませんでした。  
(修正が反映されたことで再現しなくなったのかもしれません。)

### 実行環境
- Jetson Nano(Jetpack 4.3 [L4T 32.3.1])
- Momo(自前バイナリ(12/31版使用))

### 実行方法
- 当該事象が発生したものと同様のコマンドを使用(sora モードの sendonly + multistream)

### 確認方法
- momoによる配信開始後放置、10時間稼働後別端末2台にてブラウザ経由でSora Laboにmulti_pubsub接続し配信確認
- momo再起動後、10時間放置し、LTE回線+iPhoneのSafariにてブラウザ経由でSora Laboに接続し配信確認

tetsu-koba, 2020-01-14:
1/12 にデプロイされたSora labo で私も再現テストを行っています。
接続時間は2800分 (46.6時間)を超えていますが、再現はしていません。良好です。

tetsu-koba, 2020-01-16:
連続配信しつつ時々他のブラウザからmulti_pubsub接続して様子をみるということをしていましたが、4日間で再現することはありませんでした。

voluntas, 2020-01-16:
@tetsu-koba ありがとうございます。6000 分くらいつながっていたようでした:-) ご協力ありがとうございました。すごく助かりました。

こちらはクローズします。

closed 2020-01-16 by voluntas