shiguredo/momo #131: P2PモードでUSBカメラの音声の再生ができませんでした

Shiono7777, 2020-01-28:
**バグを説明してください**
P2Pモードで映像と音声を利用しようとしたが、映像も音声もうまく再生されなかった

**再現する**
./momo --resolution VGA --fixed-resolution --port 8082 tes

**予想される行動**
映像も音声もうまく再生されない

**カメラ/マイク**
[webrtc_momo _log.txt](https://github.com/shiguredo/momo/files/4121219/webrtc_momo._log.txt)
カメラは ロジクール C922
マイクはロジクール C922の内蔵マイク

lsusb のコマンド結果を貼ってください。
Bus 002 Device 005: ID 174c:55aa ASMedia Technology Inc. Name: ASM1051E SATA 6Gb/s bridge, ASM1053E SATA 6Gb/s bridge, ASM1153 SATA 3Gb/s bridge, ASM1153E SATA 6Gb/s bridge
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 046d:085c Logitech, Inc. C922 Pro Stream Webcam
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

**自前バイナリ/提供バイナリ**

momo-2020.1_raspbian-buster_armv7

**--log-level 1 の出力ログ**

txt ファイルにて添付してください。

**追加の情報**
Raspberry pi 3 ModeB+でmomo-19.09.2_raspbian-buster_armv7では動いていました。
今回はRaspberry pi 4 Model B  4GBを使用しています。

voluntas, 2020-01-28:
@torikizi Raspberry Pi 4 で動作確認おねがいできますか？

Shiono7777, 2020-01-28:
Raspberry pi 4 Model B  4GBで動かしています。OSはRaspbian 10.2 です。VP8で接続しています。 オーディオデバイスの優先度$ cat /proc/asound/modules 0 snd_usb_audio 1 snd_bcm2835 よろしくお願いいたします。  Windows 10 版のメールから送信 差出人: voluntas送信日時: 2020年1月28日 17:27宛先: shiguredo/momoCC: Shiono7777; Author件名: Re: [shiguredo/momo] P2PモードでUSBカメラの音声の再生ができませんでした (#131) @torikizi Raspberry Pi 4 で動作確認おねがいできますか？

voluntas, 2020-01-28:
@Shiono7777 おそらく間違いの投稿がありましたので、削除しておきました。

お手数ですが apt update & apt upgrade をしているかどうかお教えいただけますか？

torikizi, 2020-01-28:
動作確認してみました。

### 実行結果
問題なく動作することが確認できました。
動作している画像はこちらです。(ラズパイ4のみで申し訳ありません)
[image]

### 実行コマンド
./momo --resolution VGA --fixed-resolution --port 8082 test

### 実行環境
完全に同一とまではいきませんが、Webカメラはロジクールを使いました。  
組み合わせは以下のようにしております。  
- ラズパイ3 + ロジクール HD Webcam C270
- ラズパイ4 + ロジクール ウェブカメラ C615

環境構築については念のため以下を参考に見直しました。
https://github.com/shiguredo/momo/blob/develop/doc/SETUP_RASPBERRY_PI.md

### 動作確認方法
テストモードで動かしてみるを参考に今回のコマンドで確認しました。
https://github.com/shiguredo/momo/blob/develop/doc/USE_TEST.md

### 所感
ラズパイの設定が3と4で異なっている場合その差が動作に影響しているかもしれません。  
　→私自身もやりがちなのですが、たとえばパッケージが古いなどです。

Shiono7777, 2020-01-29:
返信ありがとうございます。ラズパイ４のＯＳはクリアーインストールをしています。（そのようにすることを推奨するとどこかで見たため）下記のコマンドでは静止画はお送りいただいたような画面になりますが、振った手を写すなどすると、追従性がものすごく悪いです。./momo --resolution VGA --fixed-resolution --port 8082 testUSBマイクの優先度は$ cat /proc/asound/modules 0 snd_usb_audio 1 snd_bcm2835で行っています。（ロジクールのマイクを使うため）これを$ cat /proc/asound/modules０snd_bcm2835１ snd_usb_audioとなるように変更すると、当然音は再生されませんが映像はスムースに動作します。音量調整も表示されます。実行された環境がUSBマイクが優先されている環境でしょうか。お手数ですがご確認をお願いいたします。 Windows 10 版のメールから送信 差出人: torikizikaeru送信日時: 2020年1月28日 23:59宛先: shiguredo/momoCC: Shiono7777; Mention件名: Re: [shiguredo/momo] P2PモードでUSBカメラの音声の再生ができませんでした (#131) 動作確認してみました。実行結果問題なく動作することが確認できました。動作している画像はこちらです。(ラズパイ4のみで申し訳ありません)実行コマンド./momo --resolution VGA --fixed-resolution --port 8082 test実行環境完全に同一とまではいきませんが、Webカメラはロジクールを使いました。組み合わせは以下のようにしております。ラズパイ3 + ロジクール HD Webcam C270ラズパイ4 + ロジクール ウェブカメラ C615環境構築については念のため以下を参考に見直しました。https://github.com/shiguredo/momo/blob/develop/doc/SETUP_RASPBERRY_PI.md動作確認方法テストモードで動かしてみるを参考に今回のコマンドで確認しました。https://github.com/shiguredo/momo/blob/develop/doc/USE_TEST.md所感ラズパイの設定が3と4で異なっている場合その差が動作に影響しているかもしれません。　→私自身もやりがちなのですが、たとえばパッケージが古いなどです。

voluntas, 2020-01-29:
@Shiono7777 メールで返信されるのはやめて、直接 GitHub Issues への記載をお願いします。投稿記事は削除させていただきます。

Shiono7777, 2020-01-29:
OSはクリーンイントールしています。（どこかのサイトで推奨されていたので）Set upも説明どうりにやっているつもりです。./momo --resolution VGA --fixed-resolution --port 8082 test上記で実行すると静止画は送っていただいたようになりますが、手を振ったりすると、スムースに追従しません。（音量調整が表示され、音はとぎれとぎれ出ます）ロジクールのマイクを使うためAUDIOの優先順位は以下のようになっています。$ cat /proc/asound/modules 0 snd_usb_audio 1 snd_bcm2835 AUDIOの優先順位を$ cat /proc/asound/modules 0 snd_bcm2835 1 snd_usb_audioのように変更すると映像はスムースに再生されます。（音量調整は表示されますが、音は出ません） よろしくお願いいたします。 Windows 10 版のメールから送信 差出人: torikizikaeru送信日時: 2020年1月28日 23:59宛先: shiguredo/momoCC: Shiono7777; Mention件名: Re: [shiguredo/momo] P2PモードでUSBカメラの音声の再生ができませんでした (#131) 動作確認してみました。実行結果問題なく動作することが確認できました。動作している画像はこちらです。(ラズパイ4のみで申し訳ありません)実行コマンド./momo --resolution VGA --fixed-resolution --port 8082 test実行環境完全に同一とまではいきませんが、Webカメラはロジクールを使いました。組み合わせは以下のようにしております。ラズパイ3 + ロジクール HD Webcam C270ラズパイ4 + ロジクール ウェブカメラ C615環境構築については念のため以下を参考に見直しました。https://github.com/shiguredo/momo/blob/develop/doc/SETUP_RASPBERRY_PI.md動作確認方法テストモードで動かしてみるを参考に今回のコマンドで確認しました。https://github.com/shiguredo/momo/blob/develop/doc/USE_TEST.md所感ラズパイの設定が3と4で異なっている場合その差が動作に影響しているかもしれません。　→私自身もやりがちなのですが、たとえばパッケージが古いなどです。

voluntas, 2020-01-29:
@Shiono7777 メールでの投稿をおやめいただけますでしょうか。GitHub Issues へ直接お願いいたします。これ以上、守っていただけないのであれば対応できかねます。

Shiono7777, 2020-01-29:
色々なれていないのでご迷惑をおかけしました。

Raspberry pi 4 Model B  4GBの
Raspbian　OSはクリーンインストールしています。

./momo --no-audio --resolution VGA --fixed-resolution --port 8082 test
で実行すると映像は出ますが、カメラの前で手を振ったりすると
スムースに追従しません。
音声も度切れ途切れで再生されます。
ロジクールC922のマイクを利用するためにAUDIOの優先順位を

$ cat /proc/asound/modules
 0 snd_usb_audio
 1 snd_bcm2835

のようにしています。
優先順位を

$ cat /proc/asound/modules
 0 snd_bcm2835
 1 snd_usb_audio

のように変更すると、映像はスムースに再生されます。
ロジクールのマイクは使えないので音声は出ません。

voluntas, 2020-01-29:
@Shiono7777 以下、ご回答ください。Issues のタイトルが実際の問題と異なったため編集させていただきました。

- apt update & apt upgrade は行っておりますか？
  - こちらはクリーンインストールとは別物です
- --no-audio オプションを指定しているにもかかわらず、音声が配信されているという状況でしょうか？
- test モードで利用しているブラウザをお教えください

Shiono7777, 2020-01-29:
こちらはクリーンインストールとは別物です
    apt-get update & apt-get upgrade はOSインストール時に行っております。

--no-audio オプションを指定しているにもかかわらず、音声が配信されているという状況でしょうか？
    --no-audio オプションでは正しく実行しております。
    --no-audio オプションをはずし、USBマイクの優先度を上げてUSBマイクの音を利用しようとした場合に問題が発生します。

test モードで利用しているブラウザをお教えください
    windows10 Google Chrome バージョン: 79.0.3945.130（Official Build） （32 ビット）

   2020/01/29 16:45に
   apt-get update & apt-get upgrade を行いました。
   ./momo --resolution VGA --fixed-resolution --port 8082 test での動作には問題がなくなりました。
  対応ありがとうございました。

voluntas, 2020-01-29:
@Shiono7777 apt-get update & apt-get upgrade にて問題が解決したとのこで、確認ありがとうございました。こちら閉じさせていただきます。

closed 2020-01-29 by voluntas