shiguredo/momo #136: ジェットソンナノラスピカムが検出されない

dftpnd, 2020-03-06:
jetson nano raspi cam not detected
[image]
[image]

tnoho, 2020-03-06:
Sorry I have not tested with Raspberry Pi Camera. Thank you for share logs with us.
I seem the camera supported only one video format "RG10". But our program is not support it.
If you want to try to use that camera. You have to add some transform programs inside capture process.
RG10 means 10bit sRGB. It conversion is not so difficult. If you can please try to support it.

The color format conversion has written in following file.
https://github.com/shiguredo/momo/blob/develop/src/rtc/scalable_track_source.cpp#L78

But use USB camera is easiest way for you. Momo supports common USB camera formats.
Your car seems so cool! I'm looking forward to add POV function in this car. Thanks.

tnoho, 2020-03-06:
Sorry we don't support CSI camera now and we don't have any plans for support it. We are only supports USB camera. It's specification. It's not bug then I close this issue.

closed 2020-03-06 by tnoho

dftpnd, 2020-03-08:
Спасибо) желаю удави вашему проекту!