shiguredo/momo #141: ラズパイ3でHD解像度で --use-native オプションをつけると映像が出ない

tetsu-koba, 2020-03-15:
**Discord ID を教えて下さい**
tetsu_koba

**バグを説明してください**

resolution が HDで --use-native をつけると映像が出てこない。
このページに書いてある /boot/config.txt の追加はしている。
https://github.com/shiguredo/momo/blob/develop/doc/SETUP_RASPBERRY_PI.md#raspberry-pi-%E3%81%A7-usb-%E3%82%AB%E3%83%A1%E3%83%A9%E5%88%A9%E7%94%A8%E6%99%82%E3%81%AB-use-native-%E3%82%92%E4%BD%BF%E3%81%A3%E3%81%A6%E3%82%82%E3%83%95%E3%83%AC%E3%83%BC%E3%83%A0%E3%83%AC%E3%83%BC%E3%83%88%E3%81%8C%E5%87%BA%E3%81%AA%E3%81%84

**再現する**

```
SIGNALING_URL=wss://....　(省略)
CHANNEL_ID=... (省略)
AUDIO_OPT='--disable-echo-cancellation --disable-auto-gain-control --disable-noise-suppression --disable-highpass-filter --disable-typing-detection'

momo $AUDIO_OPT --log-level 2 --video-device /dev/video0 --use-native --resolution HD --framerate 30 sora --auto --video-codec H264 --audio-codec OPUS --audio-bitrate 96 --video-bitrate 2000 $SIGNALING_URL $CHANNEL_ID
```

5回試して5回とも発生した。

**期待する動作**

--resolution HD を指定したときに、 --use-native の有無にかかわらず配信ができる。

**実際の動作**

webrtc の接続はされるが、映像が流れない。
webrtc_logs_0 のログを見るとSendFrame の行が無い。

resolution を VGAにすると --use-native をつけても映像が出て、正しく配信できる。
また、--use-native をつけなければ resolution がHDでも配信できる。

**カメラ/マイク**

IMX317センサーの4K webcam
ELP-USB4KHDR01-BL36

```
$ lsusb
Bus 001 Device 004: ID 05a3:0317 ARC International 
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

**自前バイナリ/提供バイナリ**

提供バイナリ
momo-2020.3.1_raspbian-buster_armv7

**--log-level 1 の出力ログ**

webrtc_logs_0　を添付します。
[webrtc_logs_0.txt](https://github.com/shiguredo/momo/files/4333886/webrtc_logs_0.txt)

**追加の情報**

この問題に関する他の情報をここに追加してください。

tetsu-koba, 2020-03-15:
ラズパイ2 + ロジクールc270カメラの組み合わせで試しましたが傾向は同じでした。ラズパイ2なので 15fps で試しましたが、
resolution HD --use-native なし --> OK
resolution HD --use-native あり --> NG
resolution VGA --use-native あり --> OK

tnoho, 2020-03-15:
すみません。まさかの手元の C920 では問題なく動くというオチでした。手持ちの C270 で試したところ再現できました。MJPEG のデコードの時点で失敗してしまっているようです。ちょっと根が深そうなので、時間をいただくかと思います。

@voluntas マニュアルの推奨コマンドで use-native は USB カメラでは外してもらうようにしばらくは誘導した方がいいかもしれません。

voluntas, 2020-03-16:
@tnoho マニュアルの推奨から USB の場合は --use-native はつけないほうが良いですって書いておきますね。

torikizi, 2021-09-23:
こちらは最新のバイナリでも再現すること確認しました。
また、↑にコメントがあるように使用する USBカメラデバイスで事象が発生しない点についても同様でした。

私の手元のロジクール C980 では再現せず、C615  では再現しました。

melpon, 2022-03-30:
ELP USB4KHDR01-MFV でも再現しました。

Buffalo の BSW200MBK でも映像が出ない問題が発生しましたが、以下のエラーが出力されていたので別の問題かもしれません（--hw-mjpeg-decoder false では映像が出ました）。
```
mmal: mmal_port_event_send: event lost on port 1,0 (buffer header callback not defined)
```

closed 2022-04-04 by melpon