shiguredo/momo #143: Ubuntu 18.04 で libcuda.so / libnvcuvid.so がイントールされていないと起動しない

hakobera, 2020-03-22:
**Discord ID を教えて下さい**

hakobera

**バグを説明してください**

Ubuntu 18.04 x86 版の momo を NVIDIA の GPU を搭載していないマシンで動作させる場合でも、NVIDIA VIDEO CODEC SDK の一部である libcuda/libnvcuvid がインストールされていないと momo が起動しない。

momo 2020.2 までは問題なく、momo 2020.3 で NVENC 対応以降に発生するようになった事象です。

**再現する**

```console
$ ./momo --help
./momo: error while loading shared libraries: libcuda.so.1: cannot open shared object file: No such file or directory
```

**期待する動作**

libcuda/libnvcuvid がインストールされてなくても momo が起動すること。

**実際の動作**

momo が起動しない。

**カメラ/マイク**

Thinkpad E495 内蔵カメラ・マイク

lsusb のコマンド結果を貼ってください。
```console
$ lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 003: ID 5986:2130 Acer, Inc 
Bus 003 Device 002: ID 8087:0025 Intel Corp. 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

**自前バイナリ/提供バイナリ**

提供バイナリ
momo-2020.3.1_ubuntu-18.04_x86_64

**--log-level 1 の出力ログ**

起動しないので、ログはありません。

**追加の情報**

Workaround として、`libnvidia-compute-435` と `libnvidia-decode-435` をインストールすることで起動することはできる。

melpon, 2020-03-23:
https://github.com/shiguredo/momo/commit/e392434dd4bf5f4654900f9202144354889bd569 で修正しました。

hakobera, 2020-03-23:
@melpon 対応ありがとうございます。該当コミットを含む `develop` ブランチをビルドし、libcuda/libnvcuvid がなくても起動することを確認しました。

問題を報告した環境にて、`test` と `ayame` モードの両方で、VP8/VP9 が動作することを確認しました。

voluntas, 2020-04-01:
問題ないとのことなので閉じる。

closed 2020-04-01 by voluntas