shiguredo/momo #144: 2プロセス起動時に2プロセス目のmomoのCPU使用率が100%を超える

optim-yusuke-aoki, 2020-03-23:
**Discord ID を教えて下さい**

optim-aoki#5807

**バグを説明してください**

momoのプロセスを2つ起動した際、2つ目に起動したものが必ずCPU使用率100%を超える

**再現する**

- Webカメラ : 2つ接続
- Ayameモードにてmomoを2プロセス起動
  - `./momo --no-audio ayame wss://ayame-lite.shiguredo.jp/signaling **********`
  - 映像ソースがかぶらないよう `--video-device` を指定する

**期待する動作**

- 1プロセス目 : 1 ~ 5%程度
- 2プロセス目 : 1 ~ 5%程度

**実際の動作**

- 1プロセス目 : 1 ~ 5%程度
- 2プロセス目 : 100%前後

**カメラ/マイク**

```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 046d:0825 Logitech, Inc. Webcam C270
Bus 001 Device 003: ID 046d:082c Logitech, Inc.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

**自前バイナリ/提供バイナリ**

自前バイナリの方はビルドしたコミットハッシュを、提供バイナリの方はバイナリのファイル名を教えてください。

momo-2020.3.1

**--log-level 1 の出力ログ**

1プロセス目
[log1.txt](https://github.com/shiguredo/momo/files/4367250/log1.txt)

2プロセス目
[log2.txt](https://github.com/shiguredo/momo/files/4367251/log2.txt)

**追加の情報**

- 一時的な解決方法
  - 起動時に `--port 0` を追加し、エフェメラルポートから自動的に選択されるようにする

- 環境
  - momoバージョン : momo 2020.3.1
  - OS : Ubuntu 18.04
- Ayameでの映像を確認時、2つのプロセスにFPSや解像度、ビットレートの差はほぼ無い
- Ayame,Soraで現象を確認 Testモードでは確認出来ませんでした
- Ayameでは、ブラウザで映像を閲覧しなくてもCPU使用率が100%を超える
  - Soraも同様と思われるが詳しくは調べられていない

voluntas, 2020-03-23:
@optim-yusuke-aoki Discord でもお伝えしたのですが、 実は --port 0 を指定することが解決策なんです。

次のリリースで --port が指定されない場合は 0 を利用するように修正を入れようと思います。

melpon, 2020-03-25:
#147 で `--port 0` をしなくても多重起動出来るようにしました

voluntas, 2020-04-01:
@optim-yusuke-aoki こちら、修正済です。develop で確認をお願いできますか？

optim-yusuke-aoki, 2020-04-01:
@voluntas ありがとうございます。只今検証したところ動作の確認が取れました事をご報告いたします。

voluntas, 2020-04-01:
@optim-yusuke-aoki 確認ありがとうございます！！！

closed 2020-04-01 by voluntas