shiguredo/momo #158: Jetson nano + PulseAudio + rootユーザでmomoを起動する方法

soudegesu, 2020-06-09:
Jetson nanoにデフォルトでインストールされているPulseAudioを使い、非ログインユーザ（root）にて、映像＋音声送受信にてmomo(`2020.6`)のプロセスをうまく起動できずに困っています。

「PulseAudioをアンインストールせずとも起動できたよ！」という方がいらっしゃれば、手順や追加でインストールしたモジュールなど、知見を共有いただければ幸いです。
 
- 再現方法

以下のコマンドを `root` ユーザで実行します。
`./momo --log-level 0 --resolution FHD --framerate 30 sora --auto --multistream --role sendrecv --video true --video-codec H264 --video-bitrate 15000 --audio true --audio-codec OPUS --audio-bitrate 128 --metadata {認証用メタデータ} SIGNALING_URL CHANNEL`

すると以下のようなログが出力され、起動に失敗します。

```
[000:176] [8399] (audio_device_impl.cc:332): Init
[000:176] [8399] (audio_device_pulse_linux.cc:1576): mainloop running!
[000:176] [8399] (audio_device_pulse_linux.cc:1310): context state cb
[000:176] [8399] (audio_device_pulse_linux.cc:1320): no state
[000:177] [8399] (audio_device_pulse_linux.cc:1310): context state cb
[000:177] [8399] (audio_device_pulse_linux.cc:1324): failed
[000:177] [8399] (audio_device_pulse_linux.cc:1610): failed to connect context, error=-1
[000:177] [8399] (audio_device_pulse_linux.cc:144): failed to initialize PulseAudio
[000:177] [8399] (audio_device_pulse_linux.cc:1693): PulseAudio terminated
[000:177] [8399] (audio_device_impl.cc:341): Audio device initialization failed.

#
# Fatal error in: ../../../_source/ubuntu-18.04_armv8/webrtc/src/media/engine/adm_helpers.cc, line 39
# last system error: 88
# Check failed: 0 == adm->Init() (0 vs. -1)
# Failed to initialize the ADM.Aborted (core dumped)
```

soudegesu, 2020-06-09:
ubuntuのログインユーザとは違い、rootユーザではPulseAudio Serverプロセスが参照できない(`pulseaudio` コマンドで `Daemon already running` が出ない) ので `./momo` コマンド実行前に  `pulseaudio -D` を実行してあげる必要がありそうです。

soudegesu, 2020-06-09:
1. `pulseaudio -D`
2 `./momo` コマンド
のセットでとりあえずmomoの起動にこぎつけました。

しかし、momo停止時に、1のpulseaudio プロセスももれなくkillされているようです。

voluntas, 2025-11-06:
pipewire を利用する仕組みに変更しました。