shiguredo/momo #239: HWデコーダ、VP9を有効にしてFHD配信を表示すると映像下部に緑色の線が入る

solax4x, 2022-01-28:
**Discord ID を教えて下さい**

sola#5653

**バグを説明してください**

オプション「--hw-mjpeg-decoder true --vp9-encoder 1 --vp9-decoder 1」をつけて端末Aでmomoを起動し、同チャネルに対して「--resolution FHD」の解像度で端末Bから接続（配信）すると端末Aに表示される端末Bからの映像の下部分に緑色の線が入ります。
端末Bの配信する解像度を「--resolution 4K」にすると再現しませんでした。端末Bの配信方法は、後述する2021.6のmomoのバイナリやSora JavaScript SDK（2021.2.0）を試しましたが、どちらもFHDであれば再現しました。

こちらのissueに類似した内容かと思います。
https://github.com/shiguredo/momo/issues/124

**再現する**

端末A
```
./momo \
    --hw-mjpeg-decoder true \
    --use-sdl \
    --vp9-encoder 1 \
    --vp9-decoder 1  \
    sora --signaling-url [URI] --channel-id [CHANNEL_ID] \
    --auto \
    --video-bit-rate 15000 \
    --audio true \
    --video true \
    --video-codec-type VP9 \
    --role sendrecv \
    --multistream true \
    --metadata [metadata]
```

端末B
```
./momo \
    --hw-mjpeg-decoder true \
    --use-sdl \
    --fixed-resolution \
    --priority RESOLUTION \
    --vp9-encoder 1 \
    --vp9-decoder 1  \
    --resolution FHD \
    sora --signaling-url [URI] --channel-id [CHANNEL_ID] \
    --auto \
    --video-bit-rate 15000 \
    --audio true \
    --video true \
    --video-codec-type VP9 \
    --role sendrecv \
    --multistream true \
    --metadata [metadata]
```

**期待する動作**

端末Bから配信された映像の下端に緑の帯がでない

**実際の動作**

何が起きたのかについての明確で簡潔な説明をお願いします。

**カメラ/マイク**

カメラ：[ELP](https://www.amazon.co.jp/4K%E3%82%AA%E3%83%BC%E3%83%88%E3%83%95%E3%82%A9%E3%83%BC%E3%82%AB%E3%82%B9Web%E3%82%AB%E3%83%A1%E3%83%A9-IMX415%E3%82%BB%E3%83%B3%E3%82%B5%E3%83%BC2160P%E5%BA%83%E8%A7%92%E3%82%AB%E3%83%A1%E3%83%A9USB-UVC%E3%82%A6%E3%82%A7%E3%83%96%E3%82%AB%E3%83%A1%E3%83%A9Windows-800%E4%B8%87%E7%94%BB%E7%B4%A04K%E3%82%A6%E3%82%A7%E3%83%96%E4%BC%9A%E8%AD%B0%E7%94%A8%E3%82%AB%E3%83%A1%E3%83%A9%E3%82%B5%E3%83%9D%E3%83%BC%E3%83%883840x2160-PC%E5%A4%96%E4%BB%98%E3%81%91%E3%82%AB%E3%83%A1%E3%83%A9USB4K02AF-KL100/dp/B08CXYHWV9/ref=sxts_rp_s1_0?cv_ct_cx=ELP&keywords=ELP&pd_rd_i=B08CXYHWV9&pd_rd_r=056db630-37b3-453b-a292-f332ae54b030&pd_rd_w=ENol7&pd_rd_wg=wKinA&pf_rd_p=e328bf01-8bb3-4e0f-ac12-9f86b5cba769&pf_rd_r=DN5FKG8T1CJCFBSG3MVA&psc=1&qid=1643368289&sr=1-1-a032f92e-b863-4500-9015-49891d8ffe2a)

**自前バイナリ/提供バイナリ**

[momo-2021.6.0_ubuntu-18.04_armv8_jetson_xavier.tar.gz](https://github.com/shiguredo/momo/releases/download/2021.6.0/momo-2021.6.0_ubuntu-18.04_armv8_jetson_xavier.tar.gz)

**--log-level 1 の出力ログ**

環境の都合上後日追記させて頂きます。

**追加の情報**

再現した端末は以下となります。
再現端末：jetson NX
jetson：4.5.1

kzoda, 2022-01-28:
@solax4x と一緒に調査していました。

**--log-level 1 の出力ログ**
について、端末Aでの事象発生日のログを添付しますので、ご確認ください。

[webrtc_logs_0.txt](https://github.com/shiguredo/momo/files/7959096/webrtc_logs_0.txt)

torikizi, 2022-03-07:
こちら対応後マージ済みのためクローズします。

closed 2022-03-07 by torikizi